bastibl/gr-ieee802-15-4 #8: Roadmap?

JayKickliter, 2015-03-24:
I'm happy to some recent contributions to this project project. Do you have some sort of a roadmap or todo list? There's a few things I'd like to contribute, but I want to make sure I'm on duplicating effort. 

Stuff I'd like to implement:

* 900 MHz BPSK PHY
* De-hardcode address data in MAC layer

bastibl, 2015-03-24:
Hi Jay, thanks for your interest! Actually there is no roadmap, but both projects would be nice to have and I'm not aware that somebody is working on them.
There were already people asking for the 900MHz PHY, so I guess there is interest.

JayKickliter, 2015-04-05:
Ok, I have a working BPSK packet sink. I've never touched c++ before, so I had translate your original sink into Julia, get it working, and translate it back to c++. The BPSK and OQPSK sinks are so similar that in theory they could be one class, whose parameters determine the symbol mapping/bits per symbol. However, that would require performing the float slicing before the block when processing an OQPSK stream. That's unless your FM OQPSK demodulation scheme can be applied to BPSK (I really don't understand how that works).

I'll submit a pull request soon.

bastibl, 2015-04-07:
That sounds great, looking forward!

bastibl, 2015-04-10:
Since this is not really an issue I will mark it as closed. Don't know if there's a nice way of having just a thread for conversation (like in a forum). I hope that's fine.
Looking forward to your pull request.

closed 2015-04-10 by bastibl