bastibl/gr-ieee802-15-4 #15: No module named foo_swig

eniteb, 2016-12-04:
Hi, I am new to gnuradio and I get the error "No module named foo_swig" when trying to run the examples.

eniteb, 2016-12-04:
I was able to get foo_swig working but now get the following error

Traceback (most recent call last):
  File "/home/research/.grc_gnuradio/ieee802_15_4_oqpsk_phy.py", line 15, in <module>
    import foo
  File "/usr/local/lib/python2.7/dist-packages/foo/__init__.py", line 45, in <module>
    from foo_swig import *
  File "/usr/local/lib/python2.7/dist-packages/foo/foo_swig.py", line 28, in <module>
    _foo_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/foo/foo_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_foo_swig', fp, pathname, description)
ImportError: /usr/local/lib/libgnuradio-foo.so: undefined symbol: _ZN2gr5blockC2ERKSsN5boost10shared_ptrINS_12io_signatureEEES6_

bastibl, 2016-12-04:
Were there any compile / build warnings when you installed gr-foo? Can you post the output of
```
ldd /usr/local/lib/libgnuradio-foo.so
```
What GNU Radio version are you using?

eniteb, 2016-12-04:
Thanks for your response. I am using gnuradio version 3.7.9 and there where no build/compile warnings 

research@ubuntu:~$ ldd /usr/local/lib/libgnuradio-foo.so
	linux-vdso.so.1 =>  (0x00007fff1455c000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007fae24d6a000)
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007fae24b44000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fae24926000)
	libgnuradio-runtime.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9 (0x00007fae2462c000)
	libgnuradio-pmt.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.9 (0x00007fae243da000)
	libuhd.so.003 => /usr/lib/x86_64-linux-gnu/libuhd.so.003 (0x00007fae2378f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fae23407000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fae230fe000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fae22ee6000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fae22b1d000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fae22915000)
	/lib64/ld-linux-x86-64.so.2 (0x0000562b06038000)
	libvolk.so.1.1 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.1 (0x00007fae2255d000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007fae22345000)
	liblog4cpp.so.5 => /usr/lib/liblog4cpp.so.5 (0x00007fae22108000)
	libboost_date_time.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.58.0 (0x00007fae21ef6000)
	libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007fae21bee000)
	libboost_serialization.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.58.0 (0x00007fae2198d000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fae21788000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fae21570000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007fae212ef000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007fae20e8d000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007fae20af8000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fae20ad8000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007fae1f021000)

bastibl, 2016-12-04:
I just pushed some changes, that hopefully fix this. It would be great if you could try the most recent version and let me know if it works for you.

eniteb, 2016-12-04:
Thanks, you are the best. Everything is fine now

closed 2016-12-04 by eniteb

sternhenri, 2017-02-07:
Hi, 

I'm running into the same issue (also with v3.7.9) when I try to boot the example apps. 

My ldd /usr/local/lib/libgnuradio-foo.so returns:
	
> linux-vdso.so.1 =>  (0x00007ffd045a0000)
> 	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007fb4b0e9a000)
> 	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007fb4b0c74000)
> 	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb4b0a56000)
> 	libgnuradio-runtime.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9 (0x00007fb4b075c000)
> 	libgnuradio-pmt.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.9 (0x00007fb4b050a000)
> 	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb4b0187000)
> 	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb4afe7e000)
> 	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb4afc68000)
> 	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb4af89e000)
> 	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fb4af696000)
> 	/lib64/ld-linux-x86-64.so.2 (0x000055b601df7000)
> 	libvolk.so.1.1 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.1 (0x00007fb4af2df000)
> 	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007fb4af0c6000)
> 	liblog4cpp.so.5 => /usr/lib/liblog4cpp.so.5 (0x00007fb4aee89000)
> 	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007fb4aec08000)

bastibl, 2017-02-08:
What do you mean? Are you seeing "No module named foo_swig" or
```
ImportError: /usr/local/lib/libgnuradio-foo.so: undefined symbol: ZN2gr5blockC2ERKSsN5boost10shared_ptrINS_12io_signatureEEES6
```
Maybe you find some information on https://www.wime-project.net/
Make sure that you are using the most recent version.

sternhenri, 2017-02-08:
Thanks everyone the quick reply! The No module error! I'll check out wime :)

sternhenri, 2017-02-08:
Thanks for*

cn0xroot, 2018-12-18:
Hi, I get the same issues:
```
ipython
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
Type "copyright", "credits" or "license" for more information.

IPython 2.4.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import foo
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-34d390fb3acc> in <module>()
----> 1 import foo

/usr/local/lib/python2.7/dist-packages/foo/__init__.py in <module>()
     43 
     44 # import swig generated symbols into the foo namespace
---> 45 from foo_swig import *
     46 from selector import *
     47 

/usr/local/lib/python2.7/dist-packages/foo/foo_swig.py in <module>()
     26                 fp.close()
     27             return _mod
---> 28     _foo_swig = swig_import_helper()
     29     del swig_import_helper
     30 else:

/usr/local/lib/python2.7/dist-packages/foo/foo_swig.py in swig_import_helper()
     22         if fp is not None:
     23             try:
---> 24                 _mod = imp.load_module('_foo_swig', fp, pathname, description)
     25             finally:
     26                 fp.close()

ImportError: /usr/local/lib/libgnuradio-foo.so: undefined symbol: _ZN2gr6filter17mmse_resampler_cc4makeEff
```

```
from foo_swig import *
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-022ca941956a> in <module>()
----> 1 from foo_swig import *

ImportError: No module named foo_swig

```
```
gnuradio-config-info -v      
3.7.12.0
```

cn0xroot, 2018-12-19:
copy that,thanks