bastibl/gr-ieee802-15-4 #16: output RSSI and raw data

ZhijingLi, 2017-01-10:
Hi,

how to output the RSSI and even the raw IQ data for each detected zigbee packet?

Thanks.

bastibl, 2017-01-10:
Unfortunately, that's not possible out of the box. Currently the IQ samples from the SDR are demodulated and then decoded. That means at the time where a frame is detected (by finding the preamble in the demodulated data) it's not IQ anymore.
You would, for example, have to introduce a second parallel stream of raw IQ data and log/dump this IQ data once a frame is found.
While typical SDRs are no calibrated device, the  RSSI can, then, be derived from the IQ data (you also have to take the gain into account).

ZhijingLi, 2017-01-10:
Thanks a lot!

closed 2017-01-10 by ZhijingLi