bastibl/gr-ieee802-15-4 #18: ImportError: No module named ieee802_15_4_swig

vinRocha, 2017-03-08:
Hello,

I'm new to GNU Radio and I find your application very interesting. I'm trying to learn more about SDR and digital communication concepts through it. However I'm having troubles to run it. I followed the instructions at the repository, and apparently I could successful install it. I'm also sure I have all the dependencies correctly installed. However when python tries to import ieee802_15_4, I get the following error:

bastibl, 2017-03-09:
please have a look at the installation instructions at
https://www.wime-project.net/installation

vinRocha, 2017-03-10:
Bastian,

Thanks for the support. I did followed the instructions in the link you sent me, however I'm still getting the same error:

[image]

Maybe it is because I'm using a GNURadio v3.7.10.1 installed from Fedora repositories (dnf install gnuradio gnuradio-devel gr-osmosdr uhd uhd-devel ...). 

I will try to install GNURadio again using PyBombs and see if I get a different outcome.

If you have other ideia about what might be causing this import problem, I would appreciate that.

Thank you very much!

bastibl, 2017-03-11:
I cannot see the resource you linked there. Generally, I would recommend to build GNU Radio with PyBombs or from source. The versions that are shipped with the distributions are sometime very outdated.
Also make sure that you have swig installed. It's needed to create the swig bindings. If you don't have swig, you should have gotten a warning when configuring the module.

vinRocha, 2017-03-16:
Bastian,
Thank you for the support. I could manage to make everything work. I probably was with my environment completely messed up. However, build GNU Radio with PyBombs did not work well for me. So I installed it from the Fedora 25 repositories. They use up-to-date pre-compiled version of GNU Radio. I just was carefully to install all the other required dependencies, such as swig and everything worked fine. Thank you for your advises.

closed 2017-03-16 by vinRocha