bastibl/gr-ieee802-15-4 #19: Beginners guide

asad91, 2017-04-25:
Hi, I am new to GNUradio and USRP can I get simple steps for implementing a client server program between a telosb mote and USRP N210. thank you.

bastibl, 2017-04-25:
Unfortunately, there is no introductory tutorial at the moment. Since you're currently trying it out, you could document your steps and make a pull request.
To get it running, you start the QPSK flowgraph and flash the firmware in the contiki folder to the node.

asad91, 2017-04-26:
Thank you for the fast response, i am trying the example but i get following warnings and error. I am unable to resolve the issue, kindly suggest me what to do and sorry for being so noob. 

<<< Welcome to GNU Radio Companion 3.7.10.1 >>>

Block paths:
	/root/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/asad/Downloads/gr-ieee802-15-4-next/examples/ieee802_15_4_CSS_PHY.grc"
>>> Done

Loading: "/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.grc"
>>> Done

Generating: '/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py'

Generating: '/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py'

Executing: /usr/bin/python2 -u /home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.HEAD-0-g929e3b32

(process:7355): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(python2:7355): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_get_unique_name: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_register_object: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(python2:7355): GLib-GIO-CRITICAL **: g_dbus_connection_get_unique_name: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Traceback (most recent call last):
  File "/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py", line 215, in <module>
    main()
  File "/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py", line 203, in main
    tb = top_block_cls()
  File "/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py", line 95, in __init__
    channels=range(1),
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/__init__.py", line 122, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 2681, in make
    return _uhd_swig.usrp_source_make(*args)
RuntimeError: 
GR-UHD detected ABI compatibility mismatch with UHD library.
GR-UHD was build against ABI: 3.11.0,
but UHD library reports ABI: 3.10.1
Suggestion: install an ABI compatible version of UHD,
or rebuild GR-UHD component against this ABI version.

bastibl, 2017-04-26:
This error is not related to this module.
```
GR-UHD detected ABI compatibility mismatch with UHD library.
GR-UHD was build against ABI: 3.11.0,
but UHD library reports ABI: 3.10.1
Suggestion: install an ABI compatible version of UHD,
or rebuild GR-UHD component against this ABI version.
```
The GNU Radio UHD source/sink was build against a version of libuhd different from the one currently installed.

asad91, 2017-04-26:
Yes you were correct i have reinstalled GNU radio but now i am stuck with a swig error, I saw this issue posted and made sure the swig is installed on my computer but still it gives me the error

[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-162-g2790b51f
Traceback (most recent call last):
  File "/home/asad/Downloads/gr-ieee802-15-4-next/examples/transceiver_OQPSK.py", line 33, in <module>
    from ieee802_15_4_oqpsk_phy import ieee802_15_4_oqpsk_phy  # grc-generated hier_block
  File "/home/asad/.grc_gnuradio/ieee802_15_4_oqpsk_phy.py", line 16, in <module>
    import ieee802_15_4
  File "/usr/local/lib/python2.7/dist-packages/ieee802_15_4/__init__.py", line 17, in <module>
    from ieee802_15_4_swig import *
  File "/usr/local/lib/python2.7/dist-packages/ieee802_15_4/ieee802_15_4_swig.py", line 28, in <module>
    _ieee802_15_4_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/ieee802_15_4/ieee802_15_4_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_ieee802_15_4_swig', fp, pathname, description)
ImportError: libgnuradio-ieee802_15_4.so: cannot open shared object file: No such file or directory

>>> Done

bastibl, 2017-04-26:
> Yes you were correct i have reinstalled GNU radio

How about reinstalling the module. Obviously the library cannot be found...
Or you ended up with multiple installations (out of which some are broken).

Since this is not about an issue with the code, but a more general help request, it would be great if you could post your questions to the GNU Radio mailing list.

asad91, 2017-04-26:
sorry for bugging you so much the issue has been resolve after reinstalling the foo and ieee 802.15.4 from scratch. Thank you so much for the help and patience.

bastibl, 2017-04-26:
Great!

closed 2017-04-26 by bastibl