bastibl/gr-ieee802-15-4 #20: Python modules go to wrong directory on Arch Linux

yschroeder, 2017-06-12:
On Arch Linux Python modules are not installed in `/usr/local/lib/python2.7/site-packages` but in `/usr/lib/python2.7/site-packages`.

 `sudo make install` will put them in the wrong path and the modules will not be available in gnuradio and return an import error. Moving them manually to the other path works fine though...

The same is true for `gr-foo` but I think posting one issue here is enough.

You should obtain the path as described here:
https://stackoverflow.com/questions/122327/how-do-i-find-the-location-of-my-python-site-packages-directory

This returns the correct path:
```
python2 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"
```

Be sure to use explicitly `python2`, as `python` binary executes Python 3 on some systems (again Arch Linux).

Otherwise your modules work great, thank you for sharing them!

yschroeder, 2017-06-12:
Well, you are actually doing something like that here:
https://github.com/bastibl/gr-ieee802-15-4/blob/next/cmake/Modules/GrPython.cmake#L109

I'm not good with cmake, so I can't help too much. But somehow this still goes wrong and ends up using the wrong path...

bastibl, 2017-06-12:
I believe the Python modules that you install yourself should be put in `/usr/local/` and the `PYTHONPATH` should be adapted. That's what most people are doing.

If you really want it directly under `/usr/`, you'll have to configure with
```
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
```

yschroeder, 2017-06-14:
I understand that own modules should go to `/usr/local/`. However, The `PYTHONPATH` environment variable is not initialized by default so the search path of python does not include `/usr/local/`.

This might be confusing for users of your module (read: me) that receive an import error after following your installation instructions. Maybe you should add a notice about configuring `PYTHONPATH` correctly?

bastibl, 2017-06-14:
I was assuming that most people will also install GNU Radio under `/usr/local/`. I wonder what you did... 
 Do you install GNU Radio and its modules in different directories? If not, your environment variables should already have been set. Did GNU Radio actually work for you?
 Apart from that,  I think that these problems are not really related to this particular module, but should be covered by GNU Radio documentation. Otherwise, you will have to patch the documentation of all GNU Radio OOT modues out there. Independent of that, I have a note on setting the environment variables. See https://www.wime-project.net/installation/#adapt-environment-variables
Anyhow, patches to the documentation are, of course, always welcome.

closed 2017-06-14 by bastibl

yschroeder, 2017-06-14:
My gnuradio lives in `/usr/lib/python2.7/site-packages/gnuradio`. I Installed it from the official Arch package repository and it works fine.

bastibl, 2017-06-14:
Really strange that the `PYTHONPATH` was your only problem. Would expect that you will have to adapt the whole environment in case of mixed installs (i.e., when GNU Radio and OOT modules are installed in different locations). For example, I wouldn't have expect that GRC finds the blocks.