bastibl/gr-ieee802-15-4 #23: Chip Values and Corrupt packets

asad91, 2017-07-22:
Hi, I want to use study the chip values of received corrupt  packets, for that i need to disable all CRC and checksum. Can someone give me a heads start, in which files should I look and from where I can get access to the chip values? Thank you.

bastibl, 2017-07-24:
It's in the packet_sink. https://github.com/bastibl/gr-ieee802-15-4/blob/40fd8099fbf3acc417972bdaea9042a48c7f8d36/lib/packet_sink.cc#L107

closed 2017-07-24 by bastibl

asad91, 2017-08-14:
Why is the chips values are in integer? Aren't they supposed to be in 32 bit binary?

bastibl, 2017-08-14:
You seem to have a fundamental misunderstanding of data types and the printed representation of data. In C/C++ there is not dedicated data type for representing binary data. Everything is binary internally. `0b11` and 3 is the same thing.

asad91, 2017-08-14:
static const unsigned int CHIP_MAPPING[] = {
					1618456172,
					1309113062,
					1826650030,
					1724778362,
					778887287,
					2061946375,
					2007919840,
					125494990,
					529027475,
					838370585,
					320833617,
					422705285,
					1368596360,
					85537272,
					139563807,
					2021988657};

So in reality each of this value should be equal to 32 bit binary number?

bastibl, 2017-08-14:
Yes.
```
In [1]: bin(422705285)
Out[1]: '0b11001001100011111100010000101'
```
but this is not really a question for the issue tracker. Please have a look at data types in C++.

asad91, 2017-08-14:
Sir, this is just a 29 bit number and according to my knowledge a chip symbol should be of 32 bits. Decimal here ranges from 31 bits to 29 bits binary numbers. So my question stands is this the correct function for chips to symbol conversion? There is another python file in utils folder with the name symbol_mapping do you have any idea about that?. Thank you.

bastibl, 2017-08-14:
Tip: `0b1` == `0b000000000001`