bastibl/gr-ieee802-15-4 #26: ImportError: No module named ieee802_15_4

hamzaehsan, 2018-01-27:
[image]
(Screenshot attached)
I'm new to gnuradio and ubuntu. I'm facing this Import error and I find confusing how to solve it. Below "Traceback (most recent call last)" is the issue. Kindly help. Thank-you!

bastibl, 2018-01-27:
[image]

closed 2018-01-27 by bastibl

yuanfangzsp, 2022-12-12:
> [image] (Screenshot attached) I'm new to gnuradio and ubuntu. I'm facing this Import error and I find confusing how to solve it. Below "Traceback (most recent call last)" is the issue. Kindly help. Thank-you!

I have the same problem now. How did you solve it? thank you！