bastibl/gr-ieee802-15-4 #30: 868MHz/915MHz band support

dgiese, 2018-03-28:
Hi, 
any chance to support the 868MHz/915MHz bands? In contrast to 2.4GHz the standard uses BPSK instead of QPSK. Also the bitrate is different (20/40 kb/s).

ffierling, 2018-03-28:
I don't have direct experience with the 802.15.4 BPSK PHY, but it appears to have the same SHR, PHR, etc. as OQPSK. It should be easy to add it to the Access Code Prefixer, or whatever we end up calling it. As to modulation and demodulation processing, that should be possible with existing blocks.

ffierling, 2018-03-28:
Would a better name for the Access Code Prefixer be the PPDU Formatter? PPDU is the acronym used in the 802.15.4 spec for the combined SHR (Preamble + SFD), PHR and the payload.

bastibl, 2018-03-31:
Implementing this PHY should be really straightforward. The main problem is that I don't have any hardware that supports it and, therefore, cannot test it. (And implementing it without hardware is boring and much more error prone.)

Do you have a particular device in mind that you want to connect to, or why are you interested in the PHY? If it is not too expensive, I could consider buying one.

closed 2018-04-17 by bastibl