bastibl/gr-ieee802-15-4 #31: PHR Prefixer

ffierling, 2018-04-02:
PHR Prefixer prepends a vector of unsigned chars to the blob of a message tuple as advertised, but also unpacks each byte of the input blob to eight bytes in the output blob. For example, given:

PHR = (1,2,3,4,5,6,7,8,9,0xa,0xb,0xc)
and an input blob = "0x55"

PHR Prefixer outputs a blob consisting of (hex dump):

01 02 03 04 05 06 07 08 09 0A 0B 0C **01 00 01 00 01 00 01 00 01**

Two questions:
1. Is that the expected and useful behaviour?
2. Nowhere is this behaviour documented, except in the source code. Is the doxygen content in this module not propagated into the UI?

bastibl, 2018-04-02:
Yes, the PHR unpacks the byte to bits before they are modulated. This has to happen somewhere and the author chose to do it here. 

I somehow fail to understand what you are up to. Because this block is only part of the CSS PHY and not the O-QPSK PHY, which seemed to be your main interest before. It makes no sense to pick any random block and question it's usefulness separately. You have to open the flow graph where it is used and see what step it is trying to implement. In the context of this flow graph, the block does as expected and implements useful behavior.

...and you are right. The block is not documented separately. But the way it is used in the example flow graph often gives a hint what it does.

ffierling, 2018-04-02:
My immediate interest is O-QPSK and 2FSK, but I'd like to write blocks that are reusable in future 802.15.4 PHYs, and fit into an overarching scheme. 

What I'm hearing is "PHR Prefixer" is a CSS PHR prefixer, and that I should name my block "SUN PHR Prefixer"?

I'm also sincerely interested in improving the documentation, but I haven't had the time yet to dive into why the doxygen content isn't making it into GRC. Any pointers would be appreciated.

bastibl, 2018-04-02:
I think this is just not done in GNU Radio. Also the upstream blocks do not seem to pull in documentation from Doxygen, or am I missing something?

ffierling, 2018-04-02:
If you open up the Documentation tab in "PHR Prefixer" you see just its key and a prototype for make():
[image]

But if you open, say "Protocol Formatter (Async)" from gr-blocks, you get this:
[image]

The information in the Documentation tab also appears if you hover your cursor over a block in the Block Tree Panel. It saves us noobs a lot of time spelunking about in source code.

bastibl, 2018-04-02:
I just pushed a commit that (I hope) adds support for Doxygen documentation.

ffierling, 2018-04-04:
Excellent!

closed 2018-04-04 by ffierling