bastibl/gr-ieee802-15-4 #32: Modify the correlation threshold

wshe1, 2018-04-14:
Hallo, Bastian.
      I want to modify ZIgBee's correlation threshold to 10 or less.I modified your code, but I can not determine whether changes correctly. I hope you can give me a correct answer. The modified code is as follows:
[image]

bastibl, 2018-04-17:
what PHY are you trying to use?

wshe1, 2018-04-17:
I will reuse the wifi to simulate the Zigbee signal of the paper's experiment. zigbee signal of the paper's experiment. zigbee needs to reduce the correlation threshold due to the analog error of the signal.  I read your code without knowing where to correct it. I hope you can tell me where to change

bastibl, 2018-04-17:
Use WiFi to simulate Zigbee? What paper? What type of ZigBee? What experiment?

Sorry, but this leads nowhere... Maybe you find this helpful:
http://www.catb.org/esr/faqs/smart-questions.html

Once you are at a stage where you have an actual question. Feel free to open a new issue.

closed 2018-04-17 by bastibl

wshe1, 2018-04-17:
paper title:WEBee Physical-Layer Cross-Technolgy Communication via Emulation.  zigbee:MicaZ CC2420.

xushanzhizhang, 2021-07-14:
hello,wshe1
Have you finished the experimental reproduction of webee?
Recently, I have read some articles about CTC, but I don't know anything about the experimental operation. I don't know where to start? I hope you can help
thinks!