bastibl/gr-ieee802-15-4 #37: PyBombs for 15.4 not found

MountainLogic, 2018-07-13:
Not sure if Pybombs problems beling here, but I was able to install 802.11, but 15.4 failed:

``
foo@bar:~/wip$ pybombs install gr-ieee-802154
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.get_recipe - ERROR - Error fetching recipe `gr-ieee-802154':
Package gr-ieee-802154 has no recipe file!
``

However 80211 succeeded:

``
:~/wip$ pybombs install gr-ieee-80211
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - INFO - No packages to install.
``

I also tried: `` pybombs install gr-ieee802-15-4`` with no joy.

Perhaps, the name has changed?

MountainLogic, 2018-07-14:
sudo sem to solve the problem, doh!

closed 2018-07-14 by MountainLogic

bastibl, 2018-07-14:
Running it with sudo is bad practice. I guess you ran it with sudo before and, therefore, directory ownership and permissions are messed up. You could try to uninstall the module and reinstall without sudo.

MountainLogic, 2018-07-27:
Bastian,
Ignore my last comment.  I had mixed Ubuntu apt, souece builds and pybombs.  That was an evil mistake that got me hopelessly conused and I've burn my box to the ground and started with a clean install.

With the clean install and following the pybombs instructions at https://www.gnuradio.org/blog/pybombs-the-what-the-how-and-the-why, 
then I run pybombs (version 2.3.2) does not include the recipe list.  gr-foo and gr-ieee-80211 are both present in the list.

I've tried 

pybombs install gr-ieee-802154 

and pybombs then reports

ERROR  - Error fetch recipe  'gr-ieee-802154':
Package gr-ieee-802154 has no recipe file!

I'm not sure if I report this to you who ever maintains pybombs??

MountainLogic, 2018-07-27:
In my .pybombs dir I was able to clone off gr-ieee-80211.lwr and make the obvious changes to create my own gr-ieee0802154.lwr files with success.  So, I assume that '15.4 just got left out of the recipes list??? 

See https://github.com/gnuradio/pybombs/issues/510

bastibl, 2018-07-29:
Looks like the 15.4 module is not one of the "endorsed" modules that are listed in gr-recipes. The non-endorsed/misc recipes are in [gr-etcetera](https://github.com/gnuradio/gr-etcetera). You can add several recipe repos to PyBombs.