bastibl/gr-ieee802-15-4 #40: TypeError in preamble sfd prefixer - running on E312 hardware

dwalexander, 2019-01-09:
I have been running the `gr-ieee802-15-4` CSS transceiver on Ettus B205-minis with great success so far, however I am now trying to use E312 radios instead. When I try to run the transceiver example file the following error occurs:
```
root@e312:~# python transceiver_CSS_USRP.py
Using Volk machine: neon_hardfp
sender started
Traceback (most recent call last):
  File "transceiver_CSS_USRP.py", line 315, in <module>
    main()
  File "transceiver_CSS_USRP.py", line 303, in main
    tb = top_block_cls()
  File "transceiver_CSS_USRP.py", line 186, in __init__
    time_gap_2=c.time_gap_2,
  File "/home/root/.grc_gnuradio/ieee802_15_4_css_phy.py", line 57, in __init__
    self.ieee802_15_4_preamble_sfd_prefixer_ii_0_0 = ieee802_15_4.preamble_sfd_prefixer_ii((preamble), (sfd), sym_per_frame)
  File "/usr/local/lib/python2.7/site-packages/ieee802_15_4/ieee802_15_4_swig.py", line 5451, in make
    return _ieee802_15_4_swig.preamble_sfd_prefixer_ii_make(*args, **kwargs)
TypeError: in method 'preamble_sfd_prefixer_ii_make', argument 1 of type 'std::vector< int,std::allocator< int > >'
```

This happens regardless of if I am attempting to use the radio hardware to transmit or just doing a loop-back test within gnuradio itself. I have tried a few things to fix the problem but I have no idea where to start as this issue is very specific to running the code on the E312 hardware - it works fine on my computer. Any guidance here would be much appreciated.

bastibl, 2019-01-09:
Hmm, unfortunately, I don't have an ARM setup. And I don't have a really good idea.

You could try setting the Preamble and SFD parameters of the CCS PHY block in the loopback flow graph manually to 
Preamble: [1] *32
SFD: [-1, 1, 1, 1, -1, 1, -1, -1, 1, -1, -1, 1, 1, 1, -1, -1]

bastibl, 2019-02-18:
Any updates with that? I bought a RPi3, so I could try test it on ARM if there's sill an issue.

dwalexander, 2019-02-18:
I tried hardcoding all variables and this did not change the error at all. Since then I have not had time to work on the project so it has been on hold, so no updates since then. Part of my thinking is that it's to do with an incorrect cross-compilation occurring - although I'm not sure.

bastibl, 2019-02-20:
I compiled GNU Radio from source on the RPI3 but was not able to reproduce the problem. Might be due to different compile options (RPI3 is 32bit, don't know about the E312) :-/
My current best guess is that the underlying problem is that the PHY uses Numpy arrays as parameters that have to be converted to std::vectors in C++ domain. For example PHR in the constructor of the PHY is not a list but a Numpy array. Maybe you could give this a try, i.e., replace things like `c.PHR` with `c.PHR.tolist()` maybe that helps.

closed 2019-06-06 by bastibl