bastibl/gr-ieee802-15-4 #41: Beginners guide

faker-or-joker, 2019-04-10:
Hi Dr.bastibl,
First of all, I'm very grateful for the code 'gr-ieee802-15-4' you contributed.Then,I hope you could help me with some questions.
1.Do I have to use TelosB sensor motes to get started with the example you gived in the contiki folder?
2.  "edit the CONTIKI variable in the Makefile to point to your contiki installation directory "you mentioned ,I am a little confused that which contiki installation directory should I point to ,is the contiki in your example ?
3.If I use cc2530 to achieve this function,what should I do?
I'm a freshman and hope to get your guidance.

bastibl, 2019-04-10:
The Contiki example was only added to show/test interoperability with an IEEE 802.15.4 node. All your question are related to Contiki. Please see the Contiki website for further information. You can use any IEEE 802.15.4 example with this module. Contiki also contains example applications for CC2530 nodes.

faker-or-joker, 2019-04-11:
Thank you for your answer and now I have a new question.When I change the target to cd2530dk，there is a error.
user@instant-contiki:~/gr-ieee802-15-4-next/contiki$ make
  CC        ../../contiki/apps/shell/shell-power.c
../../contiki/apps/shell/shell-power.c:111: warning 85: in function process_thread_shell_power_process unreferenced function argument : 'ev'
../../contiki/apps/shell/shell-power.c:111: warning 85: in function process_thread_shell_power_process unreferenced function argument : 'data'
../../contiki/apps/shell/shell-power.c:132: warning 85: in function process_thread_shell_energy_process unreferenced function argument : 'ev'
../../contiki/apps/shell/shell-power.c:132: warning 85: in function process_thread_shell_energy_process unreferenced function argument : 'data'
../../contiki/apps/shell/shell-power.c:150: warning 112: function 'snprintf' implicit declaration
../../contiki/apps/shell/shell-power.c:150: error 101: too many parameters 
make: *** [obj_cc2530dk/shell-power.rel] Error 1

bastibl, 2019-04-11:
I just tested the example with the most recent version of Contiki. It required some minor adaptions, which I just pushed. However, I currently do not have the hardware to test it. Anyway, I did not experience the error you posted. It is very likely related to the CC2530 target.
You can debug the example for this platform, but there is really no point in getting this particular test application to run. Just use any trivial Contiki example (with Rime if you want to use it).

closed 2019-06-06 by bastibl