bastibl/gr-ieee802-15-4 #44: Error running .grc files.

r00tb3, 2019-08-05:
I've installed `GNUradio` and `IEEE-802-154` using Pybombs.

GNURadio Version:- `3.7.10`

I'm receiving the following errors when I open all the files from the examples folder.

transceiver_OQPSK.grc
```
gr-ieee802-15-4/examples/transceiver_OQPSK.grc:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
/gr-ieee802-15-4/examples/transceiver_CSS_loopback.grc"
Block key "ieee802_15_4_css_phy" not found
```

ieee802_15_4_CSS_PHY.grc
```
Loading: "/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc"
Error: /gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:1:10:ERROR:NAMESPACE:NS_ERR_QNAME: Failed to parse QName 'options:'
/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:2:14:ERROR:NAMESPACE:NS_ERR_QNAME: Failed to parse QName 'parameters:'
/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:3:5:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute parameters:
/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:3:5:FATAL:PARSER:ERR_SPACE_REQUIRED: attributes construct error
/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:3:5:FATAL:PARSER:ERR_GT_REQUIRED: Couldn't find end of Start Tag options: line 1
/gr-ieee802-15-4/examples/ieee802_15_4_CSS_PHY.grc:3:5:FATAL:PARSER:ERR_DOCUMENT_END: Extra content at the end of the document

```
What am I missing here?

Is my installation broken?

bastibl, 2019-08-05:
Please do not spam. You wrote me email privately, opened a thread with a similar question in an unrelated repository, and now you open another issue...
I already responded in this issue: 
https://github.com/bastibl/gr-ieee802-11/issues/184
If there are still problems, please continue in this thread.

r00tb3, 2019-08-05:
I've installed this module using pybombs.

If I install manually I get compilation errors and it doesn't work.

I've tried building this module and the 802.15.4 using the maint-3.7 and maint-3.8 and master branch but it fails and none of it works.

So I finally decided to go with the PyBOMBS way.

Installed GNURadio, gr-foo, gr-ieee-80211, gr-ieee-802154 using PyBOMBS and now facing the above error stated in this thread.

bastibl, 2019-08-05:
Your log suggests that gr-ieee802-15-4 was installed in the system root. I don't think that PyBombs put it there (actually no one should ever put it there). Please use the flow graphs from your PyBombs src dir. It also has the repository checked out. Also make sure you only have one installation of GNU Radio on your system.

r00tb3, 2019-08-05:
Okay, indeed I was loading the OQPSK and CSS_PHY from a wrong folder.

Now I've loaded it from the PyBOMBS folder but the files are read-only.

bastibl, 2019-08-05:
What does that mean? GNU Radio flow graphs have no permission. You probably cannot write into the directory where the flow graph is stored (maybe you ran pybombs as root or something like that). In that case change permissions of the directory. If that's not the case, what exactly are your permission problems.

r00tb3, 2019-08-05:
Okay, I chmod' ed to 777 and now the grc files are loaded correctly.

But the CSS_Loopback still says CSS_PHY is missing.

bastibl, 2019-08-05:
You have to open the PHY flow graph, build it, and reload the blocks or reopen GRC.

r00tb3, 2019-08-05:
> You have to open the PHY flow graph, build it, and reload the blocks or reopen GRC.

Thanks for all the help @bastibl .

I know I troubled you a lot.

Thanks again.

Chao.

closed 2019-08-05 by r00tb3