bastibl/gr-ieee802-15-4 #46: Unable to receive ieee802-15-4 test messages with AVR RZUSBSTICK

drtyhlpr, 2019-12-05:
Hi, thanks for the great project.

I use a b210 on debian 10.2 with gnuradio 3.7

Minimal changes were required to run the example (FE Corrections needs bool values now)
If I run the transceiver_CSS_USRP I can verify that he generated Message Strobes are shown in wireshark.

Now I am simply trying to receive the generated ieee802-15-4 packets on a second computer. I am using a Atmel RZ RAVEN via killerbee wireshark to monitor for incoming packets. I have verified that other ieee802-15-4 packets (from other iot devices) are received and monitored by the RZ RAVEN.

No matter what I try (change gains, change samplerate, switch channels etc.) I was not possible for me to detect any packet not even a broken one on the second computer using the mentioned  Atmel RZ RAVEN. The red led is blinking with send interval and no further errors etc. from gnuradio and or the b210. volk profile is ready, cpu governors on performance...

Any ideas?

ps. sadly I have no 2nd usrp/sdr hardware ready for double verification

bastibl, 2019-12-06:
Changing the gain and the channel is fine, however, you should not change the sample rate. That won't work. Sounds very good what you already did and tried. I'm not sure what goes wrong here...
I assume there are no errors. I've recently seen that the TX LED of the wrong port was blinking. Maybe try different ports to attach the antenna. A gain of ~75% of max should work. Maybe you can test again the different channels with this setting. Also make sure that you didn't do any accidental modifications of the flow graph, i.e., revert to the version in git.

drtyhlpr, 2019-12-06:
ill double check thanks!

drtyhlpr, 2019-12-07:
Hi, I gave up. No packets and I had limited time and reduced sleep.

Whats the correct sample rate btw. ? I see a pdf documentation of the module with screenshots and a 4M sample rate. The latest maint-3.7 has a default sample rate of 1M set. (I am 99,99% i fetched and used the latest pull). (PHY has a sample rate of 4M that put me to additional confusion).

what network based usrp hardware have you mainly used for the project btw?

thanks!

bastibl, 2019-12-08:
The IEEE 802.15.4 standard supports many physical layers, including the CCS PHY and the 2.4GHz O-QPSK PHY, which are supported by this module. The O-QPSK PHY is by far the most popular and the one supported by XBee. The O-QPSK flowgraph in this module uses (and always used) a sample rate of 4MHz. Please use the unmodified `transceiver_OQPSK` flowgraph to communicate with the module.

drtyhlpr, 2019-12-08:
I fetched the latest maint-3.7 just right now. I open the transceiver_CSS_USRP with GnuRadio 3.7.13.4 and the samp_rate is set to 1M.

bastibl, 2019-12-09:
Yes, but, again, Xbee doesn't support the CCS PHY... Did you read my previous comment?

drtyhlpr, 2019-12-09:
yeah sorry seems I got it wrong. I thought QPSK was part of the 802.11 wifi part (only) and I only tried the CSS.

closed 2019-12-09 by drtyhlpr