bastibl/gr-ieee802-15-4 #48: Different tx rx frequency

gmartina, 2020-05-04:
Hi,

As I can see in your video, you use the same frequency for both tx and rx.

[image]
 
I want to use broadcast, but i want rx and tx to be independent. Could this work with different rx and tx frequencies?

thank you

bastibl, 2020-05-04:
You seem to misunderstand some things. What this block shows is a "RIME Channel." You can think of it as a port in TCP/IP. It has nothing to do with frequencies. Frequencies are set in the USRP source/sink blocks. They can be set independently. I'd assume this is what you actually want.

gmartina, 2020-05-04:
Yes, I meant to set different frequencies in the hardware blocks, like sink: 2.4 GHz and source: 2.3 GHz, and in the other transceiver, the opposite: source: 2.4 GHz and sink: 2.3 GHz. This way tx and rx would be independent even using broadcast mode, right?

Ive been thinking about using Reliable Unicast Channel, but this forces me to write the address in every data i have to send, if i understand well.

On the other hand, i would like to know if a ping to the socket should work. 

(what I want to do is a simple data link between two development kits. So they can comunicate with the sockets)

Sorry for my poor network knowledge.

thank you

bastibl, 2020-05-04:
I see, that sounds about right. Just change the frequency in UHD source/sink blocks manually and use, for example, reliable unicast. You just have to put in similar ports at sender and receiver. The rest is dealt with by the block. That means you just put in the payload and it will add the necessary headers. 

Pinging, however, is a different story. At least if you mean the pings that you do with the ping utility on the command line, which sends ICMP messages. This would require connecting the GNU Radio flowgraph to the Linux TCP/IP stack. This is possible and there is an example in my gr-ieee802-11 module on how to do that. But it is (1) rather complex and hardly possible without a good understanding of networking (2) not a good idea in the first place, since IEEE 802.15.4 packets are rather small (max 127 byte) and TCP/IP add way too much overhead. 6lowPAN would be the better alternative here.

closed 2020-05-07 by gmartina