bastibl/gr-ieee802-15-4 #49: Logical Link Control question

gmartina, 2020-05-13:
Hi again Bastian,

I have tested two similar setups, one with gr-802-11 and the other with gr-802-15-4.

I use iperf v1 tool to test the speed. 802.11 works fine, you can see the Logical Link Control layer.

But 802.15.4 doesnt make iperf work. Can you tell me why? cant it put the logical link control layer itself?  is there something i am missing? 

[image]
[image]

bastibl, 2020-05-13:
There is no example for using 15-4 with TUN/TAP and it is not trivial. I'm not sure what you did, what I should see in the Wireshark screenshots, and what you mean with "15-4 did not make iperf work"...

gmartina, 2020-05-13:
I am not using TUN/TAP at all, in neither of them. Just gnuradio PDU sockets.

I use iperf with the folloing commands:

 iperf -c 127.0.0.1  -i 1 -u –p 2000 -t 60 -l 1200 -b 1M
 iperf -s -i 1 –u –p 5000

bastibl, 2020-05-13:
IEEE 802.15.4 has a maximum frame size of 127 byte (everything including headers). May that be the problem?

gmartina, 2020-05-13:
In that case, I cannot understand why I can make it to send more than 200 bytes of data.

bastibl, 2020-05-13:
What PHY are you using?

gmartina, 2020-05-14:
gr-ieee802-15-4/examples/transceiver_OQPSK.grc    (maint-3.7)

bastibl, 2020-05-14:
hmm, looks like this implementation allows a maximum frame size of 255, while the standard says it's 127 :-) It uses a Byte for the frame size. Actually one bit of the byte is reserved according to the standard.

[image]

closed 2020-08-12 by bastibl