bastibl/gr-ieee802-15-4 #50: 'ieee802_15_4' has no attribute 'rime_stack'

darrenyoung23, 2020-08-11:
Hi Bastian,

Trying to runtransceiver_OQPSK however I'm getting following error 

File "transceiver_OQPSK.py", line 134, in __init__
    self.ieee802_15_4_rime_stack_0 = ieee802_15_4.rime_stack([129], [131], [132], [23,42])
AttributeError: module 'ieee802_15_4' has no attribute 'rime_stack'

 Not sure why this is the case, is there something i am missing?

bastibl, 2020-08-12:
Make sure that:
- you have swig installed, such that Python bindings for C++ blocks are generated in the first place
- the Python module is found, i.e., it's in your PYTHONPATH
- the shared library (libgnuradio-ieee802154.so) is in your LD_LIBRARY_PATH

darrenyoung23, 2020-08-18:
Thanks, I installed swig, seems libgnuradio-ieee802154.so is missing.

darrenyoung23, 2020-08-18:
Thanks, I installed swig. seems this error is gone now. however I'm getting following error now.
return _runtime_swig.top_block_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

darrenyoung23, 2020-08-18:
running with gnuradio-config-info -v 3.8.1.0 , wondering whether this is a gnuradio thing?

bastibl, 2020-08-19:
So you use GNU Radio v3.8.1.0 with the maint-3.8 branch of this module?
Please make sure that you use the unmodified flowgraphs, i.e., git should show a clean repository. I just tested it with the GNU Radio version and it works here.

darrenyoung23, 2020-08-19:
You are correct, it's work well with the unmodified flow graph.

closed 2020-08-19 by darrenyoung23

buttercannfly, 2020-12-28:
> Make sure that:
> 
> * you have swig installed, such that Python bindings for C++ blocks are generated in the first place
> * the Python module is found, i.e., it's in your PYTHONPATH
> * the shared library (libgnuradio-ieee802154.so) is in your LD_LIBRARY_PATH

I did the three things but that didnt work though,I have swig installed by 
```
sudo apt install swig
```
then I can import ieee802_15_4 in my python,
my LD_LIBRARY_PATH has a libgnuradio-ieee802_15_4.so,
it seems it has to work now,but it didn't.