bastibl/gr-ieee802-15-4 #52: NameError(i is not defined)

buttercannfly, 2020-12-29:
```
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.
Generate Error: (NameError("'i' is not defined"), '<%\n    all_options = [ value, option1, option2, option3, option4 ][:int(num_opts)]\n    all_labels = [ label0, label1, label2, label3, label4 ][:int(num_opts)]\n%>\\\n# Create the options list\n% if int(num_opts):\nself._${id}_options = (\\\n% for ch in all_options:\n${ch}, \\\n% endfor\n)\n% else:\nself._${id}_options = ${options}\n% endif\n# Create the labels list\n% if int(num_opts):\nself._${id}_labels = (\\\n% for lbl in all_labels:\n% if lbl:\n${lbl}, \\\n% else:\nself._${id}_options[${i}], \\\n% endif\n% endfor\n)\n% elif labels:\nself._${id}_labels = ${labels}\n% else:\nself._${id}_map(str, self._${id}_options)\n% endif\n# Create the combo box\n% if widget == \'combo_box\':\n<%\n    win = \'self._%s_tool_bar\'%id\n%>\\\n${win} = Qt.QToolBar(self)\n${win}.addWidget(Qt.QLabel(${(label if (len(label) - 2 > 0) else repr(id))} + ": "))\nself._${id}_combo_box = Qt.QComboBox()\n${win}.addWidget(self._${id}_combo_box)\nfor _label in self._${id}_labels: self._${id}_combo_box.addItem(_label)\nself._${id}_callback = lambda i: Qt.QMetaObject.invokeMethod(self._${id}_combo_box, "setCurrentIndex", Qt.Q_ARG("int", self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_combo_box.currentIndexChanged.connect(\n    lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n# Create the radio buttons\n% if widget == \'radio_buttons\':\n<%\n    win = \'self._%s_group_box\'%id\n%>\\\n${win} = Qt.QGroupBox(${(label if (len(label) - 2 > 0) else repr(id))} + ": ")\nself._${id}_box = ${orient}()\nclass variable_chooser_button_group(Qt.QButtonGroup):\n    def __init__(self, parent=None):\n        Qt.QButtonGroup.__init__(self, parent)\n    @pyqtSlot(int)\n    def updateButtonChecked(self, button_id):\n        self.button(button_id).setChecked(True)\nself._${id}_button_group = variable_chooser_button_group()\n${win}.setLayout(self._${id}_box)\nfor i, _label in enumerate(self._${id}_labels):\n    radio_button = Qt.QRadioButton(_label)\n    self._${id}_box.addWidget(radio_button)\n    self._${id}_button_group.addButton(radio_button, i)\nself._${id}_callback = lambda i: Qt.QMetaObject.invokeMethod(self._${id}_button_group, "updateButtonChecked", Qt.Q_ARG("int", self._${id}_options.index(i)))\nself._${id}_callback(self.${id})\nself._${id}_button_group.buttonClicked[int].connect(\n    lambda i: self.set_${id}(self._${id}_options[i]))\n% endif\n${gui_hint() % win}')
>>> Failure
```
I installed the gr-ieee-802154 from pybombs and configured the environment,but it didn't work

bastibl, 2020-12-30:
AFAIS, this is not related to this module but was an issue with GNU Radio: https://github.com/gnuradio/gnuradio/commit/154b2b73f22cc74940a536dbcc36f55ef636c3f7
Please try updating GNU Radio.

buttercannfly, 2020-12-30:
ok,thanks for your help.
Gnuradio is strange as I installed it by Pybombs just now,its the 3.8.2 version,but It continues to report this error.
all in all,Thank you again.

closed 2020-12-30 by buttercannfly

bastibl, 2020-12-30:
Make sure that there are no left-overs from other installations. I'm not sure how PyBombs works in detail, but you might have a block path configured somewhere in your home, which points to an older installation that had this broken template.

buttercannfly, 2020-12-30:
```global_blocks_path = /usr/share/gnuradio/grc/blocks
local_blocks_path = /home/zwk/share/gnuradio/grc/blocks
```
the paths are right(I configured these two paths),and I installed pybombs and gnuradio in a fresh machine.
I checked this commit before and I find that my code is the same.
Thank you bro,I'd need to go for gnuradio for some help.

buttercannfly, 2020-12-30:
```
global_blocks_path = /usr/share/gnuradio/grc/blocks
local_blocks_path = /home/zwk/share/gnuradio/grc/blocks
```
the paths are right(I configured these two paths),and I installed pybombs and gnuradio in a fresh machine.
I checked this commit before and I find that my code is the same.
Thank you bro,I'd need to go for gnuradio for some help.

bastibl, 2020-12-30:
From what you are writing, it sounds like you'd have installed GNU Radio through your package manager and the module through PyBombs. This migh cause problems.

It looks like the global blocks path points to an installation from the package manager. Maybe this version comes with the broken tempalte. Look for `qtgui_chooser.block.yml` inside those directories and see whether they have the bug.

buttercannfly, 2020-12-31:
I have finally finished this bug,thanks for your kind.
Here is my false installation steps:
```
pybombs install gnuradio
sudo apt install gnuradio
```
it has conflicts between these two gnuradios.
Then the true steps by pybombs are
```
pybombs install gnuradio gr-foo gr-ieee-802-154 gr-ieee-802-11
source prefix-3.8/setup_env.sh
```