bastibl/gr-ieee802-15-4 #53: Not able to send packets from SDR (gnuradio) with Length more than 57 bytes.

uriasat, 2021-01-11:
Hi,

I have downloaded and working with Gnuradio Version:3.8. I have modified the transceiver_OQPSK flowgraph and made it a  transmitting radio for sending packets (attached image of flowgraph). It is working fine and we are able to send packets successfully from transmitting radio to another SDR or Openmote. 

But I am facing an issue while sending packets with lengths of more than 57 bytes. By exploring code, I have found that code is always attaching 11 bytes in mac.cc & 4 bytes of the header along with the payload. Suppose if we have a payload of length 10 bytes, SDR will transmit a total of 25 bytes. We are able to validate this length on the receiving side. So coming back to the issue, when I am sending packets with more than 57 bytes altogether, the radio is giving an error and not transmitting anything.

Can you please guide me on how to overcome this issue? I am stuck for long and not able to move forward as I am new to gnuradio. Can you please suggest in which coding file should I look into resolving the issue?

any help will be much appreciated.

H/W: USRP X310
OS: UBUNTU 20.04
GNU radio version: 3.8
_**_Error:
sched: <block foo_packet_pad2 (8)> is requesting more input data
  than we can provide.
  ninput_items_required = 8448
  max_possible_items_available = 8191
  If this is a filter, consider reducing the number of tap_**_

[image]

uriasat, 2021-01-11:
@bastibl Sir !

bastibl, 2021-01-12:
The default size of some buffers was too small to contain full-sized frames. I just pushed a fix. Let me know if it works for you.

uriasat, 2021-01-24:
Hi, sorry for getting back late.

it worked actually. Thanks a lot!

closed 2021-01-25 by bastibl