bastibl/gr-ieee802-15-4 #54: Transceiver_USRP.grc cannot work well, it receive UUUU in command

qw-Zhang, 2021-04-08:
Hi Bastian,

The PHY block is generated normally. And the simulation without USRP works well too. But the transceiver_USRP.grc cannot work well, it shows the "UUUU..." in command.

I consider it maybe caused by one PC run transmitter and receiver at the same time. Can I separate the transceiver USRP.grc to one Tx.grc and one Rx.grc and run them at two PC, like gr-ieee 802-11?

bastibl, 2021-04-13:
The 'U's are probably not the problem. I just noticed that the PHY uses `packet_len` to mark the start of a frame, while the USRP Sink uses `pdu_length`. If you adapt this, the 'U's might disappear. (Let me know if it works, then I'll push a fix.)
But, anyways, your actual issue might be that the USRP might operate in half-duplex mode (i.e., doesn't send and receive at the same time). If you're device is full-duplex-capable, you might have to adapt the parameters (e.g., antennas).

As you mentioned, you can also just use two SDRs, which should work out-of-the-box. I'd change MAC source address and RIME address for the other node, to be able to differentiate them.

qw-Zhang, 2021-04-13:
Thanks for your reply.
I have revised the parameter "TSB tag name" of USRP Sink block and regenerate OQPSK_PHY with same "packet_len". But I also receive the "UUUU....". I use the two USRP2 as transmitter and receiver. I don't know how to change the MAC source address and RIME address of the two nodes.
gr-ieee 802.15.4 version: maint-3.7 
gnuraio version: 3.7.13
This is the flow graph which I used:
[image]
And this is the output of gnuradio
[image]

bastibl, 2021-04-13:
This could be a networking issues (from a VM, a switch, etc). Try disabling the USRP source (replace with a vector source that only outputs one sample and then stops). Then check if you still see the 'U's.
If this also doesn't work, please try the maint-3.8 branch with GNU Radio 3.8.

qw-Zhang, 2021-04-14:
Finally, I upgrade gnuradio to version3.8. Everything looks wok well. Thanks for your contribution!

bastibl, 2021-04-14:
Great. Happy it worked!

closed 2021-04-14 by bastibl