bastibl/gr-ieee802-15-4 #55: Problem with receiving packets in transceiver_OQPSK .grc

diyalaisabar, 2021-05-20:
Hi, 

I am having problems with receiving of packets in transceiver_OQPSK .grc. A high sample rate does not capture any traffic at all, so the .pcap file is empty.  A low sample rate (2M-4M) does capture packets, however, I get the error message: MAC: frame to short. Dopping! I do see some packets in the.pcap file, but they are all "Malformed".  On some occasions I also receive an error message saying that the crc is wrong. I do not understand what the problem is since everything worked fine when I hade the "message" block disabled or when I try to do simple things with the hardware and the associated block , such as capture some traffic into a file or listen to the radio. 

Hardware: HackRF One
Frequency: 2.405 GHz
Branch: maint-3.8 
OS: Kali Linux 
GNU version: 3.8.2 
Communication protocol: 6LoWPAN 

Thank you in advance! 

[image]

bastibl, 2021-05-25:
You cannot just change the sample rate. The transceiver is built for 4MHz. Any other sample rate will not work with ZigBee devices. The packets that you see in the PCAP are valid frames that passed the CRC check. I assume that they are displayed as malformed because the 6LoWPAN is not used or not set up.

diyalaisabar, 2021-06-04:
I did try with 4MHz also but no packets are captured unfortunately. Do you mean as set up in Wireshark?

bastibl, 2021-06-04:
I never decoded 6LoWPAN and just know that something is malformed, I cannot help you much. Sorry. But googling 6LoWPAN and Wireshark, it looks like this dissector might require a manual setup. Also check if the payload is tried to be decoded as 6LoWPAN, in the first place. If data shows up in Wireshark, it passed the CRC check. It's very likely valid and a matter of dissecting it.

closed 2021-06-04 by bastibl