bastibl/gr-ieee802-15-4 #57: PHY layer doesn't work for Tx

JRocket789, 2022-02-24:
I've been struggling to get the Tx portion of the PHY block working for a few days now. I've followed the instructions and have left the PHY block mostly intact. I'm running GNU Radio 3.8 and I think the problem lies with the Chunks to Symbols block. For some reason, when the dimension is set to 16, the transceiver_OQPSK starts, but then instantly dies with a return code -11. When I reduce the dimension to 8 or lower, it runs just fine, but it's not creating the right spreading code. 

[image]

[image]

Hopefully there is something small and stupid that I'm missing.

bastibl, 2022-02-25:
I ran into this two days ago... It is because GNU Radio broke the Chunks to Symbols block. You'd have to revert this commit https://github.com/gnuradio/gnuradio/commit/f6b3295ae29e3ee3c2c8fc3c6738ee4561ca194b#diff-14678b5ce7ce2e320e87cebf64f2fce4283d246f8a74aadd791f418692ffd0d8L111-L113
Maybe also file a PR/Issue at GNU Radio.

JRocket789, 2022-02-28:
So I tried to revert to the commit using 'git revert f6b3295', but I kept running into issues with git trying to merge files and it inevitably wouldn't work. This was honestly probably due to my git ignorance. So I just went to that commit and downloaded the zip file associated with that date and installed gnuradio from that source code. I re-built gr-foo and gr-ieee802-15-4 and was able to get the blocks working, but the issue I was having still persists. 

Everything else is running the exact same, but the transceiver dies after a short amount of time after I run it.

Honestly, now it could definitely have been from trying to build from source instead of installing from a repo and reverting the commit. It should just be as simple as going into my cloned repo and using 'git revert f6b3295', yes?

bastibl, 2022-03-01:
There were two changes to the chunks to symbols block that you have to revert in reverse order. So you first have to revert https://github.com/gnuradio/gnuradio/commit/70a27c0a0a75dc8640dc318f2ca470de81935340 and then the other commit. This worked for me without problems.

If you download the state where this commit was just applied. You still have the unwanted commit in there. If reverting doesn't work, just download an earlier version.

Again, it would be nice, if you could report this upstream.

JRocket789, 2022-03-01:
I tried that as well, but ended up with a slew of other issues. Since the issue is with the chunks to symbols block, I'll report this upstream to GNURadio and just take your word that the revert fixes is (hopefully they'll fix the block as well). 

I noticed you have a chunks to symbols in your 802.11 package. I haven't tried using that yet, but once I get everything on my end fix, I'm going to try and use that block to see if it works. Have you had luck using your chunks2sym block vs the GR one?

bastibl, 2022-03-03:
You probably have some left-overs from previous attempts. The transceiver worked for years and years. And just reverting worked fine. Just uninstall and start from scratch. Remove all artifacts from earlier installations and recompile GNU Radio and the OOT modules.

cclamb, 2022-04-08:
Hey ya'll, I have posted this bug with the GR team: https://github.com/gnuradio/gnuradio/issues/5758

If you have anything to add or correct, please do so.

bastibl, 2022-11-02:
The upstream bug got fixed. There is a `maint-3.10` branch now. I'll close this for now. Let me know, if there are still issues.

closed 2022-11-02 by bastibl