bastibl/gr-ieee802-15-4 #59: USRP device in half duplex communication

nitinshivaraman, 2022-04-05:
Hi @bastibl 

Thanks for porting the IEEE 802.15.4 for Gnuradio. I am beginner using gnuradio and usrp. I was able to successfully transceiver_OQPSK.grc on the maint-3.8 branch. However, as per the original paper, I'm using a telosB mote and a USRP B210 to communicate. I'm able to receive messages from telosB on USRP B210 (RX2 port) while I'm unable to send any message from USRP (TX/RX) and receive on telosB. It appears that the message is successfully sent out of the PHY block when observed on the QT frequency sink (signal is around -30dB) but not sent from the antenna. I have attached the block diagram used.

Also, I'm trying to implement an existing paper to detect and correct collisions through preamble correlation (E.g. Zigzag decoding-http://ccr.sigcomm.org/online/files/p159-gollakotaA.pdf, RnR decoding-https://saifullah.eng.wayne.edu/rnr.pdf). I'm using packet_sink file for this as I'm using OQPSK PHY. In this regard, could you advise on how I can get the time difference between symbols - say two preambles from a collided packet? Can it be obtained from the clock recovery module?

Thanks in advance.

[image]
[image]

nitinshivaraman, 2022-04-14:
Thanks for the response. Adding of packet pad 2 did not make a difference. TelosB is firmware appears to be working well as I'm able to send messages to USRP and receive it. The telosB is able to receive message from other telosB but not from USRP. Is the Tx issue due to the chunks to symbols block like the earlier issue? I have installed GNURadio 3.8 using pybombs and the changes for chunks to symbols were done on the main branch. Am I missing something else? Please advise.

bastibl, 2022-04-14:
The question is if the frames sent by the USRP are similar to the ones sent by the other TelosB. Maybe it's a different MAC or RIME network/address/channel  and, therefore, gets filtered.

nitinshivaraman, 2022-04-14:
Yes, USRP sender to TelosB receiver is not working. However, to debug further, I tried USRP sender to USRP receiver, which doesn't work as well. Essentially, the transmit from the PHY doesn't seem to work. I also tried different USRP devices to ensure it is not the device problem.

bastibl, 2022-04-14:
The 3.8 branch of GNU Radio does not include the broken commits to chunks to symbols. You can also check this with the loopback configuration w/o hardware. I'd check some obvious things (antenna, gain, etc).

closed 2022-11-02 by bastibl