bastibl/gr-ieee802-15-4 #60: maint-3.9 doesn't seem to build with apt installed GR 3.9

cclamb, 2022-04-08:
Ubuntu 20.04.4 LTS, GR 3.9.5 installed via APT from ppa:gnuradio/gnuradio-releases-3.9. When running cmake (i.e. from gr-ieee802-15-4, mkdir build, cd build, cmake ../) the cmake step fails:

```
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "ieee802_15_4_python" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:241 (pybind11_add_module)
  python/bindings/CMakeLists.txt:59 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "ieee802_15_4_python" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:241 (pybind11_add_module)
  python/bindings/CMakeLists.txt:59 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:53 (add_library):
  Target "gnuradio-ieee802_15_4" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at lib/CMakeLists.txt:53 (add_library):
  Target "gnuradio-ieee802_15_4" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

```

Seems to build fine using pybombs GR 3.9.

cclamb, 2022-04-08:
I have some additional info from the cmake process; this will be the same as that with gr-foo I suspect:

```
-- Checking for module 'sndfile'
   Called from: [6]     /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake
                [5]     /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake
                [4]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/FindSNDFILE.cmake
                [3]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/gnuradio-blocksConfig.cmake
                [2]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake
                [1]     /home/cclamb/Work/gr-foo_ieee/gr-foo/CMakeLists.txt
--   No package 'sndfile' found
   Called from: [6]     /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake
                [5]     /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake
                [4]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/FindSNDFILE.cmake
                [3]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/gnuradio-blocksConfig.cmake
                [2]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake
                [1]     /home/cclamb/Work/gr-foo_ieee/gr-foo/CMakeLists.txt
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
   Called from: [6]     /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake
                [5]     /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake
                [4]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/FindSNDFILE.cmake
                [3]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/gnuradio-blocksConfig.cmake
                [2]     /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake
                [1]     /home/cclamb/Work/gr-foo_ieee/gr-foo/CMakeLists.txt
```

Perhaps an issue with the distributed 3.9.5 GR?

bastibl, 2022-04-11:
Same issue was reported here and later closed: https://github.com/bastibl/gr-ieee802-11/issues/333
I believe this is a problem of the PPA. This bug probably happens for all OOTs, i.e., it's the GNU Radio installation that has an issue. None of my modules uses libsndfile.