bastibl/gr-ieee802-15-4 #63: Chunks to symbols mapping

OxotaKpenkoe, 2022-06-28:
Hi!
First of all i want to thank you for this great work!
my question is about symbol table in "chunks to symbols" block. 
y there are some difference with ieee 802.15.4 standart? 
[image]
for example symbol # 14  mapped to this sequence in this block:
(1-1j), (-1+1j), (1+1j), (-1-1j), (-1-1j), (1+1j), (-1+1j), (-1+1j), (-1-1j), (1-1j), (-1-1j), (1-1j), (1+1j), (1-1j), (1+1j), (-1+1j)
but in ieee it is:
  10         01         01      10        00         00        01       11        01       11       10      11       10        00       11        00
so, it must be somethink like:
(1-1j), (-1+1j), (-1+1j), (1-1j), (-1-1j), (-1-1j), (-1+1j), (1+1j), (-1+1j), (1+1j), (1-1j), (1+1j), (1-1j), (-1-1j), (1+1j), (-1-1j)

bastibl, 2022-06-28:
IIRC, this is due to the strange bit ordering. 14 = 0b1110 -> reversed 0b0111 = 7. The sequence of number 14 is at position 7.

OxotaKpenkoe, 2022-06-28:
thank you so much!

closed 2022-06-28 by OxotaKpenkoe