bastibl/gr-ieee802-15-4 #64: Wrong crc, and wireshark can not work

lic9898, 2022-11-01:
Hi, when I ran the transceiver_CSS_loopback project on Gnuradio 3.10.4, I found a list of message.
[image]
I didn't change your graph.
[image]
Then I tried to catch the frame to compare tx frames and rx frames but couldn't launch wireshark. The only message is "wireshark: You can't specify both a live capture and a capture file to be read." But I created the fifo and write a bash file according to your tutorial:
$mkfifo /tmp/zigbee.pcap
$wireshark -k /tmp/zigbee.pcap
Could anyone help me ?

bastibl, 2022-11-02:
I just worked on the GNU Radio 3.10 branch and also noticed this bug. I didn't implement the CSS Phy and don't use it myself. You could switch to the O-QPSK Phy or need to debug the issue. In the latter case, a PR would be very much appreciated.

lic9898, 2022-11-03:
OK, thank you. But I don't know what PR is. Could you give me its full name?

bastibl, 2022-11-03:
I meant that if you find a solution (i.e. a fix for the bug) you should open a Pull Request (PR) here on GitHub.

lic9898, 2022-11-03:
Fine...I hope so. Thank you for your help

closed 2023-10-19 by bastibl