bastibl/gr-ieee802-15-4 #65: transceiver_OQPSK fails to transmit

dfa327, 2022-12-02:
Hello,
  Love the tools you've made and thank you for doing so :)  I'm having an issue with the transceiver_OQPSK.grc.  I get a seg fault when I uncomment the Message Strobe in GNURADIO 3.10.1.  I've narrowed it down a bit and if I send more than 8 bytes directly in to the IEEE802.15.4 OQPSK PHY block it throws the seg fault.  Can you please help me?  I've adjusted the shared memory to 8g and ran the volk profiler.  The RX works perfectly for 46 bytes packets by the way :)  So this is purely a Tx issue.  Please share any troubleshooting tips.  
  
  
[image]

9 byte test Fails:
[image]
My input looks like this:
[image]

Thank you again:)

bastibl, 2022-12-02:
GNU Radio 3.10 had a bug in one block that is used in the transmitter. You probably need a more recent version. See https://github.com/bastibl/gr-ieee802-15-4/pull/58#issuecomment-1299591824

dfa327, 2022-12-02:
Whoa!  You are fast!!!!!!  I think I found what you're talking about.  It isn't fixed in 3.10.1 nor 3.10.4, but it is in main branch :(  

https://github.com/gnuradio/gnuradio/pull/6223/commits/795fd438c16efdd5d33bd8e3900ee47d45baeca8

Thanks again for all your hard work!!!  Very much appreciated!

closed 2022-12-02 by dfa327