bastibl/gr-ieee802-15-4 #66: RuntimeError: attempt to set_msg_handler() on bad input message port!

remydemy, 2023-02-08:
Hei,

I'm trying to run the transceiver_CSS_loopback.py example. But keep on having the same error when I try to execute the file.

```
Traceback (most recent call last):
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 382, in <module>
    main()
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 360, in main
    tb = top_block_cls()
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 213, in __init__
    self.ieee802_15_4_rime_stack_0 = ieee802_15_4.rime_stack([129], [131], [132], [23,42])
RuntimeError: attempt to set_msg_handler() on bad input message port!
double free or corruption (!prev)

>>> Done (return code -6)

```

I have gnuradio, gr-foo and also gr-ieee802-15-4 all on version 3.9 and haven't changed anything in the example. This error also occurs if I try to run transceiver_CSS_USRP.py

Does anyone suspect what I did wrong?
Thanks for any help!

Best regards,
Emily

bastibl, 2023-02-08:
Unfortunately, the CSS PHY is broken for quite some time. I think the initial contributor lost interest and I don't know of anyone using it. Maybe you could use the more popular O-QPSK PHY?

remydemy, 2023-02-08:
Even when I try to use the O-QPSK PHY I still get the same error:

```
Traceback (most recent call last):
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_OQPSK.py", line 284, in <module>
    main()
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_OQPSK.py", line 262, in main
    tb = top_block_cls()
  File "/home/emily/Desktop/Thesis/gr-ieee802-15-4/examples/transceiver_OQPSK.py", line 171, in __init__
    self.ieee802_15_4_rime_stack_0 = ieee802_15_4.rime_stack([129], [131], [132], [23,42])
RuntimeError: attempt to set_msg_handler() on bad input message port!

>>> Done (return code -11)
```

I'm using the hack rf one and gr-osmocom, but that should not be the problem.

bastibl, 2023-02-09:
I just tried it with the maint-3.10 branch it works without problems. I also checked for differences between 3.9 and 3.10 that could cause this, but didn't find any. Any chances you could just give 3.10 a try? Also please check for any remaining artifact from previous installation attempts (e.g., python modules that are still in you path etc.).

remydemy, 2023-02-09:
So I removed everything that had to do with gnuradio and installed gnuradio, gr-foo and gr-ieee802154 all from the maint-3.10 branch. The problem now seems to be, that the block for O-QPSK PHY is not found. I then proceeded to uninstall and reinstall gr-ieee802154, because maybe something in the installation went wrong, but I remain to have the same problem.

[image]

Am I missing something or do I need anything else for it to work properly?
Out of curiosity I checked if the phy block in the transceiver_CSS_USRP is missing, and this block is actually missing as well. But I don't know if this says anything.

remydemy, 2023-02-09:
I again removed everything and went back to maint-3.9. For some reason the O-QPSK block is not missing anymore and I can now run it, but it still doesn't work. I get the following output when I execute it.

```
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.4.0.HEAD-0-g5fac246b
Using HackRF One with firmware 2021.03.1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.8.0
built-in sink types: uhd hackrf soapy redpitaya file 
Using HackRF One with firmware 2021.03.1
sender started
gr::log :INFO: foo_packet_pad20 - set_min_output_buffer on block 11 to 65536
gr::log :INFO: tagged_stream_multiply_length0 - set_min_output_buffer on block 15 to 65536
gr::log :INFO: float_to_complex0 - set_min_output_buffer on block 22 to 65536
Failed to start RX streaming (-6)

>>> Done (return code -11)

```

Could there be some special reason why it can not start RX streaming?

bastibl, 2023-02-09:
The O-QPSK block is a hierarchical block. Open it in GRC, click build, and re-open GRC or reload block from the GUI.
Not sure where the failed to start RX streaming error comes from, but I don't think that it has anything to do with this GNU Radio module.