bastibl/gr-ieee802-15-4 #67: No module named 'foo'

yuanfangzsp, 2023-03-09:
Hello, I encountered the following problem when running ieee802_15_4_OQPSK_PHY.grc module：

Generating: '/root/.grc_gnuradio/ieee802_15_4_oqpsk_phy.py'

Executing: /usr/bin/python3 -u /root/.grc_gnuradio/ieee802_15_4_oqpsk_phy.py

Traceback (most recent call last):
  File "/root/.grc_gnuradio/ieee802_15_4_oqpsk_phy.py", line 20, in <module>
    import foo
ModuleNotFoundError: No module named 'foo'

But my foo module is already installed。
Can you give me some help？

bastibl, 2023-03-09:
You have to check where the Python module got installed and make sure that this is found by the python interpreter, for example by setting the PYTHONPATH accordingly.

yuanfangzsp, 2023-03-15:
Can you recommend any code-related papers? I would like to further understand through the paper, thank you！

bastibl, 2023-03-31:
The above error means that the Python module was not found. If your GRC cannot find the blocks that's a different problem. If you didn't install them in a default location, you'll have to tell GNU Radio Companion where to find them. https://www.wime-project.net/installation/#adapt-environment-variables

closed 2023-10-19 by bastibl