bastibl/gr-ieee802-15-4 #68: Pybombs Install Issue

altonbill, 2023-04-05:
Hello, I am trying to install gr-ieee-15-4 via pybombs but am running into this error. From what I can tell it has something to do with the newer versions of gnuradio. Is there a way to specify an earlier release with pybombs like GR3.8?

```
bill3@node:~$ pybombs install gr-ieee-802154
[INFO] Prefix Python version is: 3.8.10
[INFO] PyBOMBS Version 2.3.5
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-ieee-802154
   |
   \- gr-foo
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-foo
warning: no common commits
remote: Enumerating objects: 1051, done.
remote: Counting objects: 100% (169/169), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 1051 (delta 85), reused 165 (delta 83), pack-reused 882
Receiving objects: 100% (1051/1051), 336.62 KiB | 2.17 MiB/s, done.
Resolving deltas: 100% (642/642), done.
From https://github.com/bastibl/gr-foo
 * [new branch]            maint-3.10 -> gr-foo/maint-3.10
 * [new branch]            maint-3.7  -> gr-foo/maint-3.7
 * [new branch]            maint-3.8  -> gr-foo/maint-3.8
 * [new branch]            maint-3.9  -> gr-foo/maint-3.9
 * [new tag]               deprecated_next_branch -> deprecated_next_branch
Cloning into 'gr-foo'...
Configuring: (100%) [============================================================================================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- PyBOMBS installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /users/bill3/prefix-3.10
-- UHD library found.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Extracting version information from git describe...
fatal: No annotated tags can describe '7557bf48f322b2347430c5ddab405ee53c57d34e'.
However, there were unannotated tags: try --tags.
-- Using install prefix: /users/bill3/prefix-3.10
-- Building for version: g7557bf48f3 / 1.1.0git
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at swig/CMakeLists.txt:36 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:50 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/users/bill3/prefix-3.10/src/gr-foo/build/CMakeFiles/CMakeOutput.log".
See also "/users/bill3/prefix-3.10/src/gr-foo/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-foo:
Configuration failed
[ERROR] Error installing package gr-foo. Aborting.

```

bastibl, 2023-04-05:
I guess PyBombs is no longer maintained. You'd either have to update the PyBombs receipt (should be simple: https://github.com/gnuradio/gr-recipes/blob/master/gr-foo.lwr#L24) or use another install method.

altonbill, 2023-04-05:
Okay, sorry if this is a naive question. I was able to install it by source, but gnuradio couldn't find the blocks. I was hoping the pybombs install would work because I couldn't find the `~/.gnuradio/config.conf` where I could point it toward where the new blocks had been built. Is there a trick to where to build the libraries from source so gnu-radio can find them?

altonbill, 2023-04-06:
Okay, it's 90% of the way there now. I've got gnu radio 3.8 to find the blocks installed from the maint-3.8 branch in Ubuntu 18. For some reason, the blocks path were in a file called grc.conf, but after adding the paths there, I can see the blocks now after I built the Phys layer blocks. However, it seems like python is picking up the ieee802_15_4 because I am seeing this message `During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/params/param.py", line 230, in evaluate
    raise Exception('Value "{}" cannot be evaluated:\n{}'.format(expr, e))
Exception: Value "ieee802_15_4.css_phy(chirp_number=4, phy_packetsize_bytes=len(text_msg)+15)" cannot be evaluated:
name 'ieee802_15_4' is not defined
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/params/param.py", line 228, in evaluate
    value = self.parent_flowgraph.evaluate(expr)
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 250, in evaluate
    return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
  File "<string>", line 1, in <module>
NameError: name 'ieee802_15_4' is not defined
`

bastibl, 2023-04-07:
This is not the actual error (*During handling of the above exception...*) I assume you either did not set the library path or python path correctly.

closed 2023-10-19 by bastibl