bastibl/gr-ieee802-15-4 #70: ieee802_15_4 has no attribute rime_stack

martinbishopx, 2023-05-05:
Hey Basti! 

I could successfully build the two hierarchical blocks. I had to adjust my library path for the python modules because it could not been found before:

`export PYTHONPATH=/usr/local/lib/python3/dist-packages/:${PYTHONPATH}`

When I am not trying to run the "transceiver_CSS_loopback.grc I am getting the following error:

```
<<< Welcome to GNU Radio Companion 3.8.5.0 >>>

Block paths:
	/home/debug/.grc_gnuradio
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.grc"
>>> Done

Generating: '/home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py

Warning: failed to XInitThreads()
Error: failed to enable real-time scheduling.
Traceback (most recent call last):
  File "/home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 374, in <module>
    main()
  File "/home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 350, in main
    tb = top_block_cls()
  File "/home/debug/gr-ieee802-15-4/examples/transceiver_CSS_loopback.py", line 205, in __init__
    self.ieee802_15_4_rime_stack_0 = ieee802_15_4.rime_stack([129], [131], [132], [23,42])
AttributeError: module 'ieee802_15_4' has no attribute 'rime_stack'

>>> Done (return code 1)
```

The blocks have been found so far. Within the code I was also not able to find the "rime_stack" attribute. Do you have an idea what went wrong? :)

[image]

martinbishopx, 2023-05-05:
I am not sure if it is important but I am using Ubuntu 20.04 in a VM. I used/built the corresponding branch (maint-3.8 ) for gnuradio compantion 3.8. The building process worked out so far. I also had to connect the lines of the CSS_PHY.grc project, but I guess, even if I made a wrong connection there, this block built should not be the problem here? 

Thanks! :)

bastibl, 2023-05-10:
Set the LD_LIBRARY_PATH to the install path of the library (not the build dir). Probably that's `/usr/local/lib`. Then do `sudo ldconfig`.

martinbishopx, 2023-05-10:
Hey Basti! Thanks for the support! I did not rebuild the project after installing swig. I did it and it worked. Should check the log next time because it mentioned swig is now being installed - wasn't before. Thanks for your help!

closed 2023-05-10 by martinbishopx