bastibl/gr-ieee802-15-4 #73: Problem Encounter While Collecting Physical Layer Data

fafrincs, 2023-10-11:
I am attempting to collect physical layer data (IQ data) using wime zigbee gnuradio project.  The setup involves two Zigbee devices connected to the laptop's ports,  and I am using XCTU software to facilitate communication between the two Zigbee devices.

[image]

However, I'm encountering an issue. I can observe messages between the two Zigbee devices in the "sensor.pcap" file using Wireshark, I'm unable to visualize this data in the QT GUI SINK; it remains empty. Additionally, when I attempt to collect physical layer data, the output of "bing.bin"  file is also empty.

My question is, how can I successfully collect IQ data after the "IEEE 802.15.4 oqpsk phy" layer block?

bastibl, 2023-10-12:
First of all remove the throttle block. It should not be used with hardware.

The time domain plot probably doesn't show anything because it is connected to the TX port, but you are probably not sending anything. I assume the flowgraph receives something but doesn't transmit itself. If you want to record the received IQ stream, you have to attach to the SDR, i.e., UHD USRP Source.

The (additional) reason why there is nothing in your output file is probably because the Tag Share block stops the stream, since there are only few samples on its second input. Remove Tags Strobe and Tags Share and you should see data in the file.

fafrincs, 2023-10-14:
[image]
I've tried various approaches, but I'm unable to obtain any IQ data from this flowgraph. I've also attached the SDR with the following configurations:
[image]

My objective is to gather physical layer data. Could you please offer any alternative suggestions for achieving this? Thank you in advance.

bastibl, 2023-10-16:
Please see my previous reply and let me know if you didn't understand it or what exactly you tried. The flowgraph in your second post hast the same issue then the one in the first, i.e., you connect the File Sink, and the time domain block to TX (and not RX).

fafrincs, 2023-10-16:
Thank you; it has been resolved.

closed 2023-10-16 by bastibl