bastibl/gr-ieee802-15-4 #74: Details about 15.4PHY

Wanghaoyu920, 2023-10-19:
I would like to see the chip sequence demodulated for each symbol, is this possible? Please help me, thanks!

bastibl, 2023-10-19:
You can add your output here:

https://github.com/bastibl/gr-ieee802-15-4/blob/maint-3.10/lib/packet_sink.cc#L114-L129

At this point, `chips` is the decoded sequence, `min_threshold` are the number of bit flips to the best matching sequence. Note that the first and last bits shouldn't be considered as indicated by the comment in the code.

Wanghaoyu920, 2023-10-19:
Thanks, if I change the source code here, do I have to recompile this OOT module? What exactly should I do?

bastibl, 2023-10-19:
do `make` and `make install` again in the build directory.

Chen-Sanyuan, 2023-12-15:
I want to know why the chip mapping is like this：CHIP_MAPPING[] = {
    1618456172, 1309113062, 1826650030, 1724778362, 778887287, 2061946375,
    2007919840, 125494990,  529027475,  838370585,  320833617, 422705285,
    1368596360, 85537272,   139563807,  2021988657
};This is different from the value defined by IEEE802.15.4

closed 2023-12-15 by bastibl