bastibl/gr-ieee802-15-4 #75: About CHIP_MAPPING

Chen-Sanyuan, 2023-12-15:
I want to know why the chip mapping is like this：CHIP_MAPPING[] = {
1618456172, 1309113062, 1826650030, 1724778362, 778887287, 2061946375,
2007919840, 125494990, 529027475, 838370585, 320833617, 422705285,
1368596360, 85537272, 139563807, 2021988657
};This is different from the value defined by IEEE802.15.4

Chen-Sanyuan, 2023-12-15:
For example, the symbol 0 is 1618456172, the binary value is 0110 0000 0111 0111 1010 1110 0110 1100. But the protocol standard is supposed to be 1101 1001 1100 0011 0101 0010 0010 1110.
[image]
Thanks a lot!

bastibl, 2023-12-15:
[Please see the comment in the code and the cited paper:](https://github.com/bastibl/gr-ieee802-15-4/blob/a210b11fed2d307ef797ea79842b54f4e8ed3dd5/lib/packet_sink.cc#L36-L47)

```
// this is the mapping between chips and symbols if we do
// a fm demodulation of the O-QPSK signal. Note that this
// is different than the O-QPSK chip sequence from the
// 802.15.4 standard since there there is a translation
// happening.
// See "CMOS RFIC Architectures for IEEE 802.15.4 Networks",
// John Notor, Anthony Caviglia, Gary Levy, for more details.
static const unsigned int CHIP_MAPPING[] = {
    1618456172, 1309113062, 1826650030, 1724778362, 778887287, 2061946375,
    2007919840, 125494990,  529027475,  838370585,  320833617, 422705285,
    1368596360, 85537272,   139563807,  2021988657
};
```

closed 2023-12-20 by bastibl

Chen-Sanyuan, 2023-12-21:
Sorry, I read the relevant articles, maybe because I know little, I did not find direct evidence of using chip mapping. I only saw a discussion that may be related to it:“In designing a ZigBee transmitter, one approach would be to implement the chip table per the IEEE 802.15.4 specification and then run the serial data stream through the encoder described above prior to FM modulation. But, the chip codes are really coding phase, not frequency, so after FM demodulation, the data stream would not represent the chip codes directly. A simpler approach is to encode the chip table itself, and use the resulting FSK spreading code table to transmit and receive data.”, but I still can’t understand why this chip mapping is used. Can you help me answer it? Thank you so much

bastibl, 2023-12-22:
I'm not sure I can provide a more comprehensive description than the cited paper, the description that you quoted, and the other material online  (e.g. https://ieeexplore.ieee.org/abstract/document/1093392).

In short, you can formulate O-QPSK in terms of MSK, i.e., these modulations are basically the same and you can turn one into the other by just mapping the bits. The receiver is a MSK receiver that uses the mapped O-QPSK chip sequence.