bastibl/gr-ieee802-15-4 #76: How  to do RF fingerprinting

lsh20049200126, 2023-12-20:
I would like to do RF fingerprinting (carrier frequency offset) for Zigbee devices based on this project
(Carrier Frequency Offset (CFO), Amplitude Error, Phase Error, DC Bias, and Synchronization
error), how can I achieve this goal

bastibl, 2023-12-20:
The Single Pole IIR Filter calculates the estimate for the CFO. The rest could be estimated in the Clock Recovery MM block. At the moment, none of this is logged or annotated to the sample data. So you would have to extend the receiver accordingly. You also have to come up with a strategy to correlated logged values with decoded frames (add tags to samples) or log absolute samples indices ([`nitems_read`](https://www.gnuradio.org/doc/doxygen/classgr_1_1block.html#a2279d1eb421203bc5b0f100a6d5dc263)) and correlate in post processing.

Chen-Sanyuan, 2023-12-21:
What are the functions of Single_Pole_IIR_Filter and Clock_Recovery_MM, and why do we need to subtract the signal that passed through Single_Pole_IIR_Filter from the demodulated signal?Thanks a lot.

bastibl, 2023-12-22:
The Single Pole IIR filter does CFO correction. Clock recovery is described here: https://wiki.gnuradio.org/index.php/Clock_Recovery_MM