bastibl/gr-ieee802-15-4 #77: IEEE 802.15.4 Block is Missing

fafrincs, 2024-03-15:
A few months ago, I worked on a project involving IEEE 802.15.4—ZigBee, and at that time, everything was functioning correctly. However, now, all of a sudden, even after building the 'ieee802_15_4_CSS_PHY.grc' and 'ieee802-15-4_OQPSK_PHY.grc' blocks, the IEEE 802.15.4 Phy block is not showing up. I don't know what the reason could be. Any help would be appreciated."

fafrincs, 2024-03-15:
[image]
[image]
I restarted GNU Radio, even the PC, and also tried on a different computer. The block is missing everywhere even after compiling the CSS and OQPSK PHY layer blocks. Have there been any changes made in the latest version?

fafrincs, 2024-03-27:
Hi,

I am still unable to solve the issue. I am using GNU Radio version 3.10.1.1 on Ubuntu 22.04.4 LTS. If you see the screenshots above, you can see that after building the PHY flowgraph, the Python files are generated. However, even after building the flowgraph, when I am trying to open the transceiver_OQPSK.grc file, the IEEE802.15.4 OQPSK PHY block is missing.

bastibl, 2024-03-27:
These files (the py and yaml file) should go to `/home/misty/.grc_gnuradio`. Not sure why they end up in the module. Maybe try copying them manually.

fafrincs, 2024-03-28:
I followed your instructions and manually copied the files using the command:
" cp /home/misty/Downloads/gr-ieee802-15-4-maint-3.10/examples/ieee802_15_4_CSS_PHY.grc    /home/misty/.grc_gnuradio" and I did this for the rest of the py and yaml files. 
However, despite copying the files, the issue still persists.

bastibl, 2024-04-02:
You are trying to run the OQPSK transceiver, which needs the OQPSK PHY (not the CSS PHY).

You are supposed to copy the generated `.py` and `.yaml` files in `~/.grc_gnuradio`.

netspooky, 2024-05-29:
Hello, I just ran into this issue and fixed it with the following steps on Ubuntu 22.04:
- Go through normal build instructions
- Open `examples/ieee802_15_4_CSS_PHY.grc` in gnuradio-companion
- Click "Generate the flow graph", this creates the yaml and python files in the examples/ directory
- Copy the generated yaml file with `sudo cp examples/ieee802_15_4_css_phy.block.yml /usr/local/share/gnuradio/grc/blocks/` 
- Reopen gnuradio-companion and open transceiver_CSS_loopback
- All the blocks should be there.
- Do the same process with `examples/ieee802_15_4_OQPSK_PHY.grc`

I'm not sure why, but my gnuradio-companion was ignoring `~/.gnuradio`. Thankfully it tells you the block paths it's reading from at start up in the terminal. I looked at both paths and the one I picked was the one with all the other generated blocks from the build.

Hope this helps!

fafrincs, 2024-05-30:
I solved the issue I was having. Sorry for informing you so late. When I installed GNU Radio on Ubuntu, the  _.grc_gnuradio_ directory was not created in _/home/misty/.grc_gnuradio_ for some reason. To fix this, I manually created the _.grc_gnuradio_ folder in that path (you need to enable the 'show hidden folders' option to see it). After reopening the flowgraph and clicking 'build', all the necessary _.py_ and _.yaml_ files were automatically generated in the _.grc_gnuradio_ directory, resolving the issue.

Thank you, @bastibl and @netspooky, for your response.

closed 2024-05-30 by bastibl