bastibl/gr-ieee802-15-4 #79: extract some RF fingerprint features

Chen-Sanyuan, 2024-07-08:
I want to extract some RF fingerprint features based on the physical layer of gr-ieee802-15-4. These RF fingerprint features are CFO, IQ imbalance, PA nonlinearity, antenna pattern. How should I achieve this goal? Thanks

bastibl, 2024-07-08:
IQ imbalance, PA nonlinearity, and antenna pattern are independent of this module. The CFO is corrected with the IIR filter, following the demodulator.

Chen-Sanyuan, 2024-07-08:
Can I use the input: phase difference of a single-pole IIR filter to get the CFO? Thanks.

Chen-Sanyuan, 2024-07-08:
Can I use the input of a single pole IIR filter, i.e. the phase difference, to get the CFO? Thanks.

Chen-Sanyuan, 2024-07-08:
Does this mean that I can estimate CFO by calculating the phase difference after processing it through the IIR filter? Thanks

bastibl, 2024-07-08:
You can regard O-QPSK as a form of FM and decode it accordingly. If you were perfectly synchronized the _average_ phase difference between consecutive samples (that's what the quadrature demodulator outputs) would be zero. If you introduce a CFO, it shifts the output of the demodulator away from zero. The bigger the CFO, the bigger the shift. The IIR filter calculates the average and later subtracts it, to compensate the CFO and get back to a signal with a zero average.

Chen-Sanyuan, 2024-07-08:
So, the result of the IIR filter is the CFO? Thanks a lot.

bastibl, 2024-07-08:
Yes, it's the estimate of the CFO in radians per sample.