bastibl/gr-ieee802-15-4 #83: Recovery of IQ Samples

irfanbabar42, 2025-08-09:
Hi,
I greatly appreciate your efforts. I am working on collecting the IQ samples of the message which is transmitted.  I used SDRAngel to send the message with IEEE 802.15.4 (OQPSK) modulation. While I am trying to capture the specific IQ samples of the received messages.

I tried to record the IQ samples from "ieee802_15_4_oqpsk_phy".  Unfortunately, it does not output the IQ samples of the message. While I can see the message is correctly shown in my Wireshark capture. 

How can I record the specific IQ samples? 
Thanks in advance.

[image]

bastibl, 2025-08-12:
This is currently not supported. You could just capture all samples from the SDR without using the decoder and decode the file afterwards. Storing only the samples that belong to a decoded frame is a non-trivial extension.

bastibl, 2025-08-12:
You'll have to study the code for the details, but very short: the IIR and Subtract remove carrier frequency offset, which results in a non-zero average after Quadrature demod. Clock Recovery outputs one sample per bit (trying to find the ideal sample points using a feedback loop) and Packet Sink despreads the DSSS bit sequences and decodes the data. And as I said, finding the the sample where the signal started is not supported and would require a non-trivial extension.